IRNAS/pulseox-hardware #5: V2 - battery

VojislavM, 2017-07-19:
LiPo battery for powering device. 
* JST-PH 2-Pin connector on the board, for battery connection.  
* LiPo charging circuit
* battery is charged over USB, and there is "hot swap" between battery and the USB power. 

The battery capacity will be determent later on. This is a low power device so the battery peak current should not be a problem.

VojislavM, 2017-07-19:
Battery needs to be rethink because it seems that the current device is powered of a standard AA battery.

VojislavM, 2017-08-07:
Because this is a portable electronic device better choice is LiPo.
4.20v = 100%
4.03v = 76%
3.86v = 52%
3.83v = 42%
3.79v = 30%
3.70v = 11%
3.6?v = 0%

LukaBanovic, 2017-09-12:
@kostko here is everything you need for charging control.

VojislavM, 2017-09-12:
PIN PA1

[image]

kostko, 2017-09-15:
So this has changed from the previous PCB? Because previously A1 was used for power management of the detector (to turn it on/off). So looking at the new PCBs, the detector is now always powered?

kostko, 2017-09-15:
Ok, @VojislavM confirmed that this is the case. Previously this was meant for reducing power usage by turning the detector off when it was not needed (but this was never actually used in the firmware).

We should benchmark battery performance to see if we will need to do any power optimizations.

kostko, 2017-09-15:
Battery voltage monitoring is now working with my development firmware. We need to find a good way to display this information on the (limited) screen. Any suggestions?

LukaBanovic, 2017-09-15:
I suggest keep the size of the "HR" and the bpm number as it is but move both down so that the SpO2 and HR numbers are aligned at the bottom, then there is some space free in the top right corner. You can add in the time and constantly display the calculated "remaining" time of the battery in xH yzM.

VojislavM, 2017-09-21:
https://learn.sparkfun.com/tutorials/battery-technologies/lithium-polymer

LukaBanovic, 2017-10-24:
I've measured the battery, when full, the voltage is 4.21V. Present BAT LO sign when the voltage drops below 3.5V.

LukaBanovic, 2017-10-24:
I suggest using screw terminal for battery pins instead of headers as a universal solution for battery mounting. Currently the pin holes on some LiPo batteries connectors are too small to fit onto header pins.